SystemRDL/systemrdl-compiler #129: Array index errors when trying to get address

shareefj, 2022-06-20:
Hi,

I'm trying to write a walker but am hitting an error when using arrays.  I've generated the following testcase:

```
addrmap CSR {
    default regwidth = 32;
    default hw = r;
    default sw = rw;

    reg PFB_CTRL {
        field {} EN[0:0] = 0;
    };

    reg PFB_REGION_ENABLE #(longint unsigned REGIONS) {
        field {} EN[REGIONS-1:0] = 0;
    };

    reg PFB_REGION_END #(longint unsigned MSB) {
        field {
        } ADDRESS[MSB:10] = 0;
    };

    reg PFG_REGION_START #(longint unsigned MSB) {
        field {
        } START[MSB:10] = 0;
    };

    regfile PFB_REGION_ADDR #(longint unsigned MSB) {
        PFG_REGION_START #(.MSB(MSB)) START @0x0;
        PFB_REGION_END   #(.MSB(MSB)) END   @0x4;
    };

    PFB_CTRL                         PFB_CTRL            @0x00;
    PFB_REGION_ENABLE #(.REGIONS(2)) PFB_REGION_ENABLE   @0x04;
    PFB_REGION_ADDR   #(.MSB(19)   ) PFB_REGION_ADDR [2] @0x10 += 8;
};
```

and am trying to get the address using:

```python
    def enter_AddressableComponent(self, node):
        self.log.debug(f"enter_AddressableComponent: name={node.inst_name}")
        self.log.debug(f"  absolute_address={node.absolute_address}")
```

but am getting a ValueError:

```
  File ".venv/lib/python3.9/site-packages/systemrdl/node.py", line 866, in absolute_address
    return self.parent.absolute_address + self.address_offset
  File ".venv/lib/python3.9/site-packages/systemrdl/node.py", line 808, in address_offset
    raise ValueError("Index of array element must be known to derive address")
ValueError: Index of array element must be known to derive address
make: *** [hjson] Error 1
```

Am I doing something wrong here?  Or perhaps the better question is how am I supposed to expand an array and get all the addresses?

shareefj, 2022-06-20:
And while I browse the code base... shouldn't the following format string have a `dim` inside it? i.e. the argument should be `empty_array_suffix: str="[{dim}]"`?

https://github.com/SystemRDL/systemrdl-compiler/blob/main/systemrdl/node.py#L740

shareefj, 2022-06-21:
I've managed to find the `unroll=False` default to RDLWalker.  Is that really meant to be false?  What is the usage model of not unrolling any arrays?

So I'm now getting my register arrays correctly parsed but I'm still having trouble with the regfile arrays.  When a regfile is unrolled, how would you go about uniquifying the register name?  Do I need to check in each `enter_Reg` whether or not the parent is a regfile and whether it is an array?

amykyta3, 2022-06-21:
It looks like you had a few questions. I will try to address them individually:

# ValueError when using absolute_address
The `Node.absolute_address` property can only be used if the node's full hierarchical lineage is known. In your case, you referenced it through a node that represented an aggregate array (without a specific known index), so it is unable to return a meaningful absolute address.

If you wish to get the address of each element in an array, then you should traverse the design with `unroll=True`. (Looks like you already figured that out)

See the documentation for more examples/details: https://systemrdl-compiler.readthedocs.io/en/stable/model_traversal.html

# empty_array_suffix's default value
The "dim" formatting variable was added in a later release. To preserve compatibility, the default value of `empty_array_suffix` was un-changed. This was intentional.

# Why is the default unroll=False?
It really depends on the use-case. When building code generators, it is often very useful to traverse the register model by visiting each unique node only once. Making this the default behavior was intentional.

# How to uniquify register names
Since SystemRDL allows complex designs that have nested hierarchies and arrays, this is probably the most common challenge of any code generator. As you have discovered, `Node.inst_name` is not guaranteed to be a globally unique identifier.
How you decide to solve this is up to you and depends on the unique constraints of what you are tying to accomplish.

The easiest way to generate names that have a high chance of being unique is to derive the name based off of the hierarchical path.
For example:
```python3
unique_inst_name = node.get_path(hier_separator="__", array_suffix="_{index:d}")
```
This is a pretty easy and robust way to make things unique, but has the downside of generating some longer names that can be cumbersome.

shareefj, 2022-06-21:
Thanks a lot for the comprehensive answer, makes sense.  On the unique name front, yeah, I'd sort of come to the conclusion that I'd need to keep track of any array and create some named prefix but `get_path()` looks a bit simpler.  Thanks.

closed 2022-06-21 by shareefj