SystemRDL/systemrdl-compiler #130: Enum parsing

markcohen32, 2022-06-28:


markcohen32, 2022-07-10:
Hi, is there an update regarding this bug?

amykyta3, 2022-07-19:
Was on vacation. Will look into it more now that I am back

markcohen32, 2022-08-07:
Hi, any new? :)

amykyta3, 2022-08-18:
I have a plan on how to fix this, but it is more intrusive than I had hoped.
The plan is to migrate away from the use of Python's `Enum` objects to represent user-defined enumerations. This would address this issue, as well as other quirks that result from this (inability to pickle register models). Instead of `Enum` classes, RDL enumerated types would be represented by more specialized class instances. A similar change would be done for RDL struct representations, but i expect they are less often used.

The biggest challenge is that I need to build a migration path that is not intrusive to the community and do my best to not break compatibility with current usages. I have a general plan for this, but still need to think through some details.

amykyta3, 2022-08-23:
See planned rework details in #132.
I hope to get this fixed in the next few weeks.

amykyta3, 2022-09-08:
Fixed in v1.24.0

closed 2022-09-08 by amykyta3