SystemRDL/systemrdl-compiler #134: Assignment of parameter value to Perl variable

markcohen32, 2022-09-06:
Hi, Is there any syntax that supports this following case?
`addrmap ADDRMAP_EXAMPLE #(longint unsigned SIZE= 2) {
  
   <% for ( $i = 0; $i < $SIZE; $i += 1 ) { %>
            sub_addrmap sub_addrmap_inst_<%=$i%>;
            sub_addrmap_inst_<%=$i%> -> hdl_path = "hdl_path_example_for_inst_<%=$i%>";
   <% } %>
};`

I'm trying to create some kind of bypass to the unsupported assignment of properties of array elements.

amykyta3, 2022-09-07:
No this is not possible.
The Perl preprocessor processes the input file prior to any RDL source parsing. RDL parameters are evaluated at the end - once the RDL source has been compiled and final expressions are being elaborated.

markcohen32, 2022-09-07:
Thanks for the quick response. Is there any plan to support dynamic assignments? Or, do we have any way to use the existing abilities to use workaround for it?

amykyta3, 2022-09-08:
I have some ideas on how to address #51, but it currently is not a priority.

closed 2022-09-10 by amykyta3