SystemRDL/systemrdl-compiler #135: Access to the parameters of the component

markcohen32, 2022-09-12:
Hi,

I've been trying to find a method/property of some components that would give me an access to the parameter definitions and values.
For instance:

```
reg reg_example #(longint unsigned REGWIDTH= 32, string DESCRIPTION = "original description")  { 
		
		regwidth = REGWIDTH;
		
		desc = DESCRIPTION;
		
		//additional content + fields ...
};
```

Instantiation:
`
reg_example #(.REGWIDTH(64),.DESCRIPTION("test description")) reg_example_inst;
`
I'd like to have access to the parameters through the RegNode component that would allow me to extract the defined parameters, their original + override values.
Does something like that exists in the current API?

Thanks

amykyta3, 2022-09-13:
The information exists, but it is not part of the public API. Here's how you would extract that information. Assume "my_node" represents an instance of "reg_example":
``` python
# Get final parameter value after any overrides
for p in my_node.inst.parameters:
    if p.name == "REGWIDTH":
        value = p.get_value()
        break

# Get original parameter value prior to any override
for p in my_node.inst.original_def.parameters:
    if p.name == "REGWIDTH":
        value = p.get_value()
        break
```

It is important for me to note - since this is not part of the publicly documented API, I cannot guarantee that this will work in the future in case I decide to change the underlying implementation.

Since I'm curious, what are you aiming to accomplish in this situation?

markcohen32, 2022-09-13:
My Intention is to create an automation that would create verilog registers corresponding to the rdl registers. And, as can be assumed from my question, I would like to use parametrization too.

markcohen32, 2022-09-13:
Another question, in case of assignment of a parameter value by another parameter - can this data be accessed too?
Continuing the previous example:

```
regfile regfile_example #(TOP_REGWIDTH_PARAM = 64) {
        reg_example #(.REGWIDTH(TOP_REGWIDTH_PARAM),.DESCRIPTION("test description")) reg_example_inst;
}
```
I'd like to have the information saying that the property regwidth is assigned by a parameter called "REGWIDTH" and this parameter is assigned by another parameter called "TOP_REGWIDTH_PARAM".

amykyta3, 2022-09-15:
The `parameters` variable will contain a list of `Parameter` objects. These have a member `expr` which is the beginning of the  un-elaborated AST (Abstract Syntax Tree) that represents the expression tree that was assigned to the value.
I suppose you can traverse it.

Links: https://github.com/SystemRDL/systemrdl-compiler/blob/main/systemrdl/core/parameter.py#L15
AST objects: https://github.com/SystemRDL/systemrdl-compiler/tree/main/systemrdl/ast

But reminder - this is well beyond the "stable API" and I guarantee I will be changing it once I start work on #58. I suspect once I complete that feature, the ability to defer parameter evaluation will more closely resemble what you actually need.

markcohen32, 2022-09-15:
When is #58 supposed to be implemented? 
Also, is there another way to approach this? I mean, querying for example the property of the component and see which parameter assigned its value?
For example - regwidth is assigned by REGWIDTH parameter. That would be helpful.

amykyta3, 2022-09-16:
I hope to start work on #58 this year, but we'll see. Keep in mind that all of these SystemRDL tools are done in my free time in the evenings. It has become a bizarre hobby project for me :laughing: 

Currently, there is no way to determine _how_ a particular value was assigned to a property since all that information is discarded during elaboration. All assignment expressions are evaluated into their final values. The only thing that remains of the abstract syntax tree are the leftover references in the parameters I mentioned earlier. Once #58 is implemented, you'll be able to mark a top-level parameter as "deferred" (or whatever I decide to call it) and then any assignments that interact with that parameter will end up preserving that part of their AST.

amykyta3, 2022-09-16:
I missed this in your prior comment:
> My Intention is to create an automation that would create verilog registers corresponding to the rdl registers

Be sure to check out my other project - [Peakrdl-regblock](https://peakrdl-regblock.readthedocs.io/en/latest/).
This is a SystemVerilog RTL generator that may do exactly what you need. It does not support dynamic parameterization yet, but it is definitely something I want to implement in the future.

markcohen32, 2022-09-19:
> I hope to start work on #58 this year, but we'll see. Keep in mind that all of these SystemRDL tools are done in my free time in the evenings. It has become a bizarre hobby project for me 😆
> 
> Currently, there is no way to determine _how_ a particular value was assigned to a property since all that information is discarded during elaboration. All assignment expressions are evaluated into their final values. The only thing that remains of the abstract syntax tree are the leftover references in the parameters I mentioned earlier. Once #58 is implemented, you'll be able to mark a top-level parameter as "deferred" (or whatever I decide to call it) and then any assignments that interact with that parameter will end up preserving that part of their AST.

This ability would be helpful. Hope you'll have the time to implement this. Traversing the AST seems to be pretty difficult and, as you said, unstable.

Anyway, I'm not sure if the following question is worth another thread, but it's also related to RDL parameters.
Is there a way to declare a parameter array with a known size and initialize it accordingly? for instance, something like that:
```

regfile example #(
    longint unsigned ARRAY_SIZE = 2,
    bit RESET_VAL[ARRAY_SIZE ]   = '{0},
   ) {
 // regfile content
};
```

amykyta3, 2022-10-04:
Unfortunately it doesn't look like SystemRDL defines a mechanism for defining auto-sized array literals.
Also, SystemRDL array parameters do not get declared with a dimension size, so you would have to do this instead:
```
bit RESET_VAL[]   = '{0, 0},
```

closed 2022-10-04 by amykyta3