SystemRDL/systemrdl-compiler #141: alignment not checked when address is explicit

j-klier, 2022-11-02:
Explicit addresses are not checked against the `alignment` property.
The 2.0 spec says that all addresses **shall** match the alignment (5.1.2.2.1)
> The alignment property defines the byte value of which the container’s instance addresses shall be a multiple.

Example:
```
addrmap ALIGNMENT4 {
    alignment = 4;
    reg {
       regwidth = 16;
       field {} SPARE[15:0] = 16'h0;
    } SPARE @0x2;
};
```

Here's a potential fix in elaborate.py
```
            if child_node.inst.addr_offset is not None:
                if (child_node.raw_address_offset % prop_alignment) != 0:
                    self.msg.error(
                        "Address offset +0x%x of component '%s' must be aligned on a 0x%xB boundary"
                        % (child_node.raw_address_offset, child_node.inst.inst_name, prop_alignment),
                        child_node.inst.inst_src_ref
                    )
                # Address is already known. Do not need to infer
                prev_node = child_node
                continue
```

amykyta3, 2022-11-03:
Thanks! I agree that this should be reported as an error.

amykyta3, 2023-01-04:
Fixed in v1.25.2

closed 2023-01-04 by amykyta3