SystemRDL/systemrdl-compiler #146: Memories to not pick up the default width

krcb197, 2022-12-17:
Consider the following memory definition

```systemrdl
addrmap memories {
        external mem {
            sw = rw;
    } mem_1_32_rw;

    addrmap wide_memory_region {
        default regwidth = 64;
        external mem {
            sw = rw;
        } mem_1_64_rw;
   };
} my_memories;
```

The SystemRDL specification says:

> 10.1 e) The default width of a register (regwidth) is 32 bits.
> 11.3.1 d) memwidth defaults to regwidth
> 11.3.1 b) mementries defaults to 1

Therefore I would expect this memory to have the following attributes:

| Memory              | Width | Entries |
| ------------------ | ------- | ------- |
| mem_1_32_rw     | 32       | 1         |
| mem_1_64_rw     | 64       | 1         |

However, this is not the case the width seems to get set to 0.

amykyta3, 2023-01-03:
The `default` assignment statement only takes effect on components that allow that property assignment inside that lexical scope (see 5.1.3.2). Since the `mem` in your example does not contain any `reg` definitions (and `regwidth` is not a valid property of `mem` components), the `regwidth` assignment does not actually have any effect.
Also since the `mem` component does not contain any registers, it is unable to derive the memory's width as per 11.3.1. 

It seems that the current implementation results in a width of 0, which I admit is a little confusing. Instead, the compiler really ought to emit an error stating that the memwidth is undefined. I will fix that.

amykyta3, 2023-01-04:
error msg added in v1.25.2

closed 2023-01-07 by amykyta3