SystemRDL/systemrdl-compiler #152: when i use peakrdl to test this generic_example .rdl,it will has"external chip_id chip_id_reg @0x0000;"error,do peakrdl dose not support the keyword of “external”？

jiangqingliu88, 2023-02-10:


amykyta3, 2023-02-10:
Are you trying to use this with the `peakrdl regblock` command? Because if so, `PeakRDL-regblock does not support external components yet.
Other commands will support it, but not the regblock generator.

See https://peakrdl-regblock.readthedocs.io/en/latest/limitations.html#external-components

I plan to implement it eventually: https://github.com/SystemRDL/PeakRDL-regblock/issues/4

jiangqingliu88, 2023-02-21:
thanks，i also have a problem，the front door is suucess access，also，i try to use the backdoor mode to access  the register of DUT ，so  aligment to hdl_path ：“top.dut.filedstroage,but i cannot use backdooraccess the filedstroage.block.register.feld.value,so, the types of data/signal  are struct result in error，the front door is suucess access。

jiangqingliu88, 2023-02-23:
Thank you, I have solved the backdoor access perfectly, your open source tool is great to help me with advanced co-design verification.

jiangqingliu88, 2023-02-23:
i I have two suggestion。The first suggestion， the struct { } reg，must add packed，for example struct paced{} reg；otherwise，the backdoor not allow access，not support the unpaked data struce。The second suggestion，Your fiedstorage storage mode is small endian mode（LSB）, so the backdoor access is in the hardware need to change to big endian mode（MSB）, or in the software to convert, We can compare it to the front door data.

amykyta3, 2023-02-23:
Backdoor access should work without issue on unpacked structs. What simulator too are you using? It may be a limitation of the simulator.