SystemRDL/systemrdl-compiler #158: SystemRDL standard future?

Risto97, 2023-03-27:
Hello,

I have a more general question about the SystemRDL standard.
I like the SystemRDL language and I think it has a lot of potential, but sadly I feel like its missing a lot of features to make it more versatile.

To give an example, I am trying to create an interconnect generator for SoC, but there are some features that I am missing in order to describe it in an elegant way. Namely:
 - Interfaces/Buses, it would be nice to have a way to describe a bus in some way, which should be a collection of signals with some additional properties to them. And then associate them to an addressmap indicating which interconnect should be used for components internally.
- Signal direction, although this can be easily added via property, it feels like its a missing feature in the standard.
- Non structural components that can hold signals, parameters, properties, something like an addrmap but with no addressable nodes inside, this could be used for describing masters on the interconnects.
- Component types passed as parameters to components

Surely I can get around by hacking and using existing features to make it work, but it doesn't feel right.
Looking at the Accellera page, it seems like the standard is abandoned? At least the working group is declared dormant, whatever that means.
Do you have maybe more insight on whether they are planning to continue developing the standard or they consider it complete?

If the standard is abandoned are there any plans on forking the standard and adding missing features if any?

amykyta3, 2023-03-31:
For a lot of the things you describe you can specify them using your own user defined properties. The spec isn't going to be able to anticipate every specific use-case, so they intentionally added a provision for user defined properties so you can extend it and define your own behavior.

My interpretation of SystemRDL is that it is better suited for describing peripherals than entire interconnects/NoCs. More specifically, SystemRDL generally only captures one monolithic address space whereas a NoC may provide multiple memory maps depending on the mode or port. Another thing I noticed is that SystemRDL usually does not attempt to impose any specifics regarding the implementation of a register map.

Regarding the spec, yes the official stance from Accellera is that the committee is dormant. That does not mean the standard is abandoned at all. It simply means they are not actively working on a new revision at this time.
For this compiler and related projects, I currently have no intent to fork the standard since that would defeat the purpose of having a standard in the first place.

If you have questions regarding the standard itself, I'd recommend checking out Accellera's forum here: https://forums.accellera.org/forum/52-systemrdl-discussion/

closed 2023-04-05 by amykyta3

kbroch-rivosinc, 2023-04-05:
It is a shame that the systemRDL WG is dormant.  User defined properties are definitely necessary but the issue with them can becomes vendor lock-in on tooling (thankfully PeakRDL is OSS 🙏 @amykyta3)  

I heard from a previous systemRDL WG member (current IP-XACT (IEEE 1685) WG member) that there use to be a give and take from IP-XACT (mostly bus definition) and systemRDL standards.  🤞 this happens in the future.

Perhaps if you choose to use systemRDL, you can use the IEEE-1685-2022 https://standards.ieee.org/ieee/1685/10583/ to help inform your user defined properties.

Risto97, 2023-04-06:
Thank you for the insight.
Well IP-XACT would be something I am looking for, only I don't understand its actual use case.
The only way I can see it being used is with a GUI because it is not meant to be written by hand, but I avoid using a GUI for this.
Another option is to write the language on top of IP-XACT, or some python bindings, but then you are reinventing the language.
There is this language I came across https://semifore.com/examples/ , which is supposed to be able to describe interconnects as they claim, but if you look at it it looks awfully similar to SystemRDL.

My view of SystemRDL was that it is meant to provide a human-writable alternative to IP-XACT.
Digging deeper into the standard I am confused to why they developed SystemRDL at all, as it has overlapping with IP-XACT, but its missing features from IP-XACT, so it feels unfinished. That said, I still think its a good language and syntax.

So how I am trying to solve the bus definition problem is by having a bus defined as a Addrmap, with signals.

```
addrmap nmi_intf #(
    intf_t INTF = intf_t'{name:"nmi_intf", ADDR_WIDTH:32, DATA_WIDTH:32, prefix:"", modport:Modport::slave},
    longint unsigned N=1
    ){
    intf;
    intf_inst = INTF;

    nmi_valid #(.W(N))                nmi_valid;
    nmi_ready #(.W(N))                nmi_ready;
    nmi_instr #(.W(1))                nmi_instr;
    nmi_addr #(.W(INTF.ADDR_WIDTH))    nmi_addr;
    nmi_wdata #(.W(INTF.DATA_WIDTH))   nmi_wdata;
    nmi_rdata #(.W(N*INTF.DATA_WIDTH))   nmi_rdata;
    nmi_wstrb #(.W(INTF.DATA_WIDTH/8)) nmi_wstrb;

 reg {name = "dummy"; field {sw = r; hw = w;} dummy[1] = 0;} dummy;
};
```
But because I need to define a memory element in Addrmap, otherwise it fails to compile (which makes sense).
I cannot use this directly in the hierarchy, so instead I use additional properties to reference to which interfaces are used in the module.

```
property sections {
    type = string;
    component = mem;
};

addrmap nmi_memory #(
    intf_t INTF = intf_t'{name:"nmi_intf", ADDR_WIDTH:32, DATA_WIDTH:32, prefix:"s_", modport:Modport::slave},
    longint unsigned MEM_SIZE   = 4096,
    longint unsigned DATA_WIDTH = INTF.DATA_WIDTH,
    longint unsigned ADDR_WIDTH = INTF.ADDR_WIDTH,

    string           SECTIONS   = ""
    )
    {
    name = "NMI memory";
    desc = "Memory module with NMI (Native Memory Interface) port";

    intfs = '{ INTF };

    mem {
        mementries = MEM_SIZE/(INTF.DATA_WIDTH/8);
        memwidth   = INTF.DATA_WIDTH;
        sections   = SECTIONS;

    } external memory;

    clk clk;
    rstn rstn;

};
```

So not exactly flawless, but still has a clean enough syntax.

kbroch-rivosinc, 2023-04-06:
> Another option is to write the language on top of IP-XACT, or some python bindings, but then you are reinventing the language.

https://github.com/olofk/ipyxact

amykyta3, 2023-04-07:
@Risto97, I think you are attempting to use SystemRDL to describe something well beyond the scope for what the language was intended for.
SystemRDL is for describing memory mapped address spaces, not for describing the structure of a CPU bus interface.
In general, the language is intentionally designed to be an abstraction _away_ from the specific details of the hardware implementation of the address space - this is why you don't see support for specific signal-level details for bus interfaces.
SystemRDL is intended as a human-writable input for tools like [PeakRDL](peakrdl.readthedocs.io) to generate peripheral register block RTL, documentation, and other collateral.

Semifore's CSRSpec is similar to SystemRDL, but instead is a proprietary language that is locked to that specific commercial tool. I dont expect that language to be too different in philosophy from SystemRDL except that it is not an open standard like RDL.

IP-XACT serves a completely different purpose from both of these - it is used as an IP interchange description format. You are correct that it includes a register description but also it lets one describe an IP's ports, source files, and many other aspects of an IP. IP-XACT is usually used as metadata to go along with an IP's RTL files, not to actually implement it.