SystemRDL/systemrdl-compiler #159: Reference produced by operations on other references

eblankenberg-trt, 2023-03-30:
Is there any mechanism for creating references that are combinational-logic products of other references?

I am seeking to implement a double-buffered register structure where propagation is controlled both by a hardware-supplied signal and the value of a software-controlled field (which self-resets on the hardware propagation event). A key component of this implementation is the ability to produce a reference which is a combinational-logic product of other references. I am having trouble implementing this. The standard seems like it should be comprehensive enough to capture this use case without additional hacks. 

From the SystemRDL documentation it is unclear whether boolean operations can be used on a reference to produce another reference. It is also unclear if signals can be driven internally by references (the spec only considers the case where the signal reference is externally driven). In SystemRDL project it appears that boolean operators always must operate on compile-time constant booleans to produce other compile-time booleans. Attempting to use references which are one-bit wires/fields (or reductions of longer references) with boolean operators produces the error "rhs/lhs is not a compatible boolean type".

I could bring in more external signals and do the combinational logic in a custom HDL wrapper but I was hoping to take advantage of the regblock-compiler as a one-stop-shop and not require additional wrapper logic.  I am aware of reduction-logic in fields but creating dummy fields for performing operations seems inelegant and I would prefer not to have a lot of fluff in generated documentation. From further experimentation concatenation seems to be only allowed on compile time constants (strings, integer literals) so even that approach may not work.

The simple example below demonstrates this use case:

```
addrmap addrmap_top {
    signal {
        signalwidth = 1;
        activehigh = true;
        sync = true;
    } external_prop_en;

    reg {
        field {
            fieldwidth = 32;
            sw = rw;
            hw = na; // Should be w if driven by internal SystemRDL logic?
        } value;
    } reg_input;

    reg {
        field {
            fieldwidth = 1;
            sw = rw;
            hw = na; // Should be w if driven by internal SystemRDL logic?
        } value;
    } reg_control;

    reg {
        field {
            fieldwidth = 32;
            sw = r;
            hw = rw;
        } value;
    } reg_buffered;

    reg_control.value->hwclr = external_prop_en;
    reg_buffered.value->next = reg_input.value;
    reg_buffered.value->we = external_prop_en && reg_control.value; // Ok if either externaL_prop_en or reg_control.value alone but operation on both to create a new reference does not work...
};
```
Thank you in advance for any assistance. Please let me know if there is a better channel for this question.

eblankenberg-trt, 2023-03-30:
Apologies, looked around a bit more and it seems that support for boolean expressions in reference assignments is not currently supported (ex. #154).

If you have suggestions for implementing the above functionality I would love to get feedback (and so I will keep the issue open for now). Feel free to close the issue.

amykyta3, 2023-03-31:
Not sure if this is what you're looking for, but be sure to check out the read/write buffered register feature I added to the regblock generator: https://peakrdl-regblock.readthedocs.io/en/latest/udps/write_buffering.html

From your example, It isn't entirely clear to me what you are trying to accomplish. But as a workaround to the boolean expression limitation, you can just as easily assign `reg_buffered.valid->we = true` to infer an input signal on the `hwif_in` bus, and then drive it from outside with the appropriate signals looped back.

eblankenberg-trt, 2023-03-31:
Yep this is exactly what I was looking for, I think write-buffering with a small amount of wrapper logic for an externally supplied trigger should work for my use case. Not sure how I missed that section of the wiki, thanks for the link (and generally the excellent open-source tools!).

closed 2023-03-31 by eblankenberg-trt