SystemRDL/systemrdl-compiler #160: Interleaved registers arrays

masics, 2023-03-31:
I have following file:
```
addrmap test {
    field my_field_t {};

    reg {
        my_field_t f1[0:0] = 0;
        my_field_t f2[15:8] = 123;
    } reg1;

    reg {
        my_field_t x[0:0] = 0; 
    } reg_array1[2] @ 0x100 += 0x20;

    reg {
        my_field_t x[0:0] = 0;
    } reg_array2[2] @ 0x110 += 0x20;
};
```

I expect reg_array1[0] at address 0x100, reg_array2[0] @ 0x110, reg_array1[1] @ 0x120 and reg_array2[1] @ 0x130.

Instead I get error:

```
error: Instance 'reg_array2' at offset +0x110:0x14F overlaps with 'reg_array1' at offset +0x100:0x13F
    } reg_array2[2] @ 0x110 += 0x20;
      ^^^^^^^^^^
fatal: Elaborate aborted due to previous errors

```

What is wrong?

amykyta3, 2023-03-31:
SystemRDL does not allow interleaved arrays.

Instead of attempting to interleave them, I would recommend grouping them using an array of regfiles to build some hierarchy. This avoids the confusing interleaving, while also grouping your concepts more concisely:

```
addrmap test {
    field my_field_t {};

    reg {
        my_field_t f1[0:0] = 0;
        my_field_t f2[15:8] = 123;
    } reg1;

    regfile {
        reg {
            my_field_t x[0:0] = 0; 
        } reg_a @ 0x00;

        reg {
            my_field_t x[0:0] = 0;
        } reg_b @ 0x10;
    } regs_array[2] @ 0x100 += 0x20;
};
```

This results in the exact same address offsets as before.

masics, 2023-03-31:
It's a workaround rather than a solution.
For example, this file cannot be solved with your proposal:
```
addrmap test {
    field my_field_t {};

    reg {
        my_field_t x[0:0] = 0;
    } reg_array1[20] @ 0x100 += 0x20;

    reg {
        my_field_t x[0:0] = 0;
    } reg2 @ 0x110;
};
```

amykyta3, 2023-04-05:
Correct. My understanding of the SystemRDL spec is that this would be prohibited since the two blocks intersect.

In general I would not recommend this type of structure anyways since it causes map uniformity issues.

masics, 2023-04-05:
Thank you.

Could you please point me to the SystemRDL spec specific section?

Best regards,
Michael Strelnikov

amykyta3, 2023-04-06:
A quick scan of the spec and I couldn't find it. I forget if this was based on something I saw in the spec, or some clarification I got from a conversation with a committee member (will have to dig through my personal emails).

Regardless, I have posted the question here: https://forums.accellera.org/topic/7513-interleaved-registers/

amykyta3, 2023-04-12:
I found the specific reason register interleaving is not supported. This was a conversation I had a while ago from a committee member. 

SystemRDL was designed to be a counterpart to IP-XACT. For this reason, the spec authors would strive to ensure that structural semantics of SystemRDL would be compatible with IP-XACT. This includes IP-XACT's array overlap rules which are described in section B.2.7-SCR7.1 of IEEE Std 1865-2014. This clause is very explicit in defining a registers range as inclusive of every element in the array:
[image]

closed 2023-04-12 by amykyta3