SystemRDL/systemrdl-compiler #166: Override parameter during elaborate, second time default parameter is used

Risto97, 2023-04-09:
Take a look at this example:

test.rdl
```
addrmap test #(
    longint unsigned N = 5
    ){
    reg {name = "dummy"; field {sw = r; hw = w;} dummy[1] = 0;} dummy;
};
```
In test code I first override parameter N with 2, then I override for another instance with 3.
I expect print to be:
N 2
N 3
```
from systemrdl import RDLCompiler

rdlc = RDLCompiler()
rdlc.compile_file("test.rdl")

node = rdlc.elaborate(
        top_def_name="test",
        inst_name="new_inst",
        parameters={'N' : 2}
        ).find_by_path("new_inst")
[print(p.name, p.get_value()) for p in node.inst.parameters]

node2 = rdlc.elaborate(
        top_def_name="test",
        inst_name="new_inst2",
        parameters={'N' : 3}
        ).find_by_path("new_inst2")
[print(p.name, p.get_value()) for p in node2.inst.parameters]

```
What I get is:
N 2
N 5

If I remove default parameter value it works correctly and I get:
N 2
N 3

Seems like a bug?

amykyta3, 2023-04-09:
Interesting. Yeah agree that doesn't look right.
I suspect this was missed since in nearly all the cases I use this, I only ever run the `elaborate()` step once. What you are doing should be supported though! Something must be leaking between elaborations.

Thanks for reporting this!

Risto97, 2023-04-09:
I am doing weird things with the compiler honestly. But some corner cases are popping up, which is good.

I feel like the problem is the _copy_for_inst() method in component.py

amykyta3, 2023-04-20:
Found and fixed the issue. Fix will be in the next release.

closed 2023-04-20 by amykyta3

Risto97, 2023-07-06:
@amykyta3 I think the issue was not completely fixed.
I am using 1.26.0 systemrdl compiler.

test.rdl
```
struct test_struct{
    longint unsigned num;
};

addrmap test #(
    test_struct STRUCT = test_struct'{num:5},
    longint unsigned some_num = STRUCT.num
    ){

    reg {name = "dummy"; field {sw = r; hw = w;} dummy[1] = 0;} dummy;
};
```

test.py
```python
from systemrdl import RDLCompiler, RDLCompileError
from systemrdl.rdltypes.user_struct import UserStruct, UserStructMeta
import os, sys

from systemrdl.core import parameter
from collections import OrderedDict

rdl_file = os.path.join(os.path.dirname(__file__), "test.rdl")

rdlc = RDLCompiler()
rdlc.compile_file(rdl_file)

# Parameter is a user struct

number = 10
struct_literal = f"test_struct'{{num: {number}}}"
struct = rdlc.eval(struct_literal)

root = rdlc.elaborate(
        top_def_name="test",
        inst_name="new_inst2",
        # parameters={'INTF' : { 'num': 32 }}
        parameters={'STRUCT' : struct}
        )

node = root.find_by_path("new_inst2")
assert node is not None

overrided_param = None
for p in node.inst.parameters:
    try:
        print(p.name, p.param_type,  p.get_value().members)
    except:
        print(p.name, p.param_type,  p.get_value())

number = 20 
struct_literal = f"test_struct'{{num: {number}}}"
struct = rdlc.eval(struct_literal)

root = rdlc.elaborate(
        top_def_name="test",
        inst_name="new_inst2",
        # parameters={'INTF' : { 'num': 32 }}
        parameters={'STRUCT' : struct}
        )

node = root.find_by_path("new_inst2")
assert node is not None

overrided_param = None
for p in node.inst.parameters:
    try:
        print(p.name, p.param_type,  p.get_value().members)
    except:
        print(p.name, p.param_type,  p.get_value())
```

stdout
```
STRUCT <class 'systemrdl.rdltypes.user_struct.test_struct'> OrderedDict([('num', 10)])
some_num <class 'int'> 10
STRUCT <class 'systemrdl.rdltypes.user_struct.test_struct'> OrderedDict([('num', 20)])
some_num <class 'int'> 5
```

As you can see in the next elaboration, the structure value is correct, but the integer value is incorrect