SystemRDL/systemrdl-compiler #169: Array stride for last element may be wrong

kjeldbeg, 2023-04-21:
I compile this example:
```
addrmap myMap {
  reg { 
    field {
      sw = rw;
      hw = r;
    } f1;
  } reg1[2] += 0x100;
  reg {
    field {
      sw = rw;
      hw = r;
    } f2;
  } reg2;
};
```
Our old RDL compiler puts reg2 at offset 0x104 but systemrdl-compiler puts reg2 at 0x200. The standard claims that += specifies the spacing of the elements. It does not state that the stride should be used as if it was the size of the last element.

Which offset for r2 is the right one?

amykyta3, 2023-04-26:
The SystemRDL spec doesn't do a great job of describing the specific behavior of this situation. However, I have had some conversations with the original spec authors that worked on the standards committee, and they mentioned that the SystemRDL spec's intention was to have a close correlation to the rules and semantics of the IEEE 1865 IP-XACT standard.

Looking at the IP-XACT standard, you'll find that placing `reg2` at offset 0x104 would cause an overlap violation according to rule SCR 7.18:
[image]

To conform to compatibility with IP-XACT and other similar standards, I interpret similar array size computation rules for the SystemRDL compiler as well. This addressing rule ends up being useful since it prevents incompatibilities in numerous other export formats as well, such as generation of C/C++ structs.

amykyta3, 2023-04-26:
From conversations I have had in the past with SystemRDL spec committee members, the semantics of SystemRDL and IP-XACT have historically been very closely intertwined. I will reach out to the RDL committee members and see if I can get more explicit clarification. I agree that I would prefer to not have to guess at the spec's intent, and would prefer to get an authoritative answer on this. There are [numerous parts of the specification that have had similar issues regarding vagueness](https://systemrdl-compiler.readthedocs.io/en/stable/dev_notes/rdl_spec_errata.html).

Are you able to share which RDL compiler makes this interpretation? Even some of the commercial compilers have been known to intentionally deviate from the spec, which is not ideal.

kjeldbeg, 2023-04-28:
Unfortunately I am not in a position where I can disclose the name of the compiler vendor.

I have access to another RDL compiler. It turns out it puts reg2 at 0x200. So, indeed a clarification is needed.

closed 2023-08-26 by amykyta3

amykyta3, 2023-10-29:
Looks like Richard Weber (IP-XACT and SystemRDL committee member) replied and confirmed the existing interpretation is preferred. See https://forums.accellera.org/topic/7529-interpretation-of-total-array-size-implicit-address-allocation/

kjeldbeg, 2023-10-30:
Great. I can use that for reference.