SystemRDL/systemrdl-compiler #170: Generate one cycle strobe signal on empty register read

m-kru, 2023-04-25:
Does anyone know, how to define in SystemRDL an empty register that can be read, and each read generates one cycle strobe signal in RTL?

amykyta3, 2023-04-26:
SystemRDL does not allow totally empty registers, but you could do something like this with a dummy field:
```
reg {
    field {
        sw=r; hw=w;
        swacc;
    } x;
} read_trigger_reg;
```

The `swacc` property on the field infers an output signal that strobes when the field is read.
The [PeakRDL-regblock RTL generator supports this](https://peakrdl-regblock.readthedocs.io/en/latest/props/field.html#swacc).

m-kru, 2023-04-26:
@amykyta3 is `hw = w` required in this case?

amykyta3, 2023-04-26:
Good point. You could assign `hw = na` instead.

m-kru, 2023-04-26:
Can I bound `swacc` to `reg` instead of `field`?

amykyta3, 2023-04-26:
Nope. swacc is a field property

m-kru, 2023-04-26:
By the way, is there any way to define a constant within the `addrmap`? I know I can parametrize elements, but this is not what I want.

amykyta3, 2023-04-27:
Besides parameters RDL doesn't really have a concept for constants. However there is a full Verilog-style preprocessor which you could use to define a macro that contains your constant.

closed 2023-05-01 by m-kru