SystemRDL/systemrdl-compiler #173: Cannot alias registers with we = true

Gragaloth, 2023-05-11:
I'm unable to generate an alias of a register where `we = true`, as I get an error complaining that the aliased copy has `we = true` when it isn't implementing logic.

```
field foo_f {
    sw = r;
    hw = rw;
    we = true;
};

reg bar_r {
    foo_f foo;
};

addrmap top {
    //Bar Register
    bar_r bar @0x100;

    //Alias to Bar
    alias bar bar_r bar2 @0x200;
};
```

```
aliasing_bug.rdl:4:5: error: Use of 'we' property on field 'foo' that does not implement storage does not make sense
    we = true;
    ^^
fatal: Elaborate aborted due to previous errors
```

The spec requires the hardware properties of the alias to be identical (10.5.1), so it would be nice if alias registers ignored this check on `we`.

amykyta3, 2023-05-11:
Agree this is a bug. Alias registers naturally do not have their own storage element, but it looks like the rule check isn't waived if it is an alias.

Thanks for finding this.

amykyta3, 2023-06-09:
Fixed in v1.26.0

closed 2023-06-09 by amykyta3