SystemRDL/systemrdl-compiler #176: Empty addrmap cause compiler to crash

kjeldbeg, 2023-05-22:
This example:

```
addrmap mymap_t {
  addrmap {
  } empty;
};
```

causes the compiler to crash.

amykyta3, 2023-06-01:
Confirmed. This would still be a compilation error since empty addrmaps are not allowed, but I see that it completely crashes during elaboration. Will fix!

amykyta3, 2023-06-01:
Will be fixed in next release. Thanks!

closed 2023-06-01 by amykyta3

Risto97, 2023-06-01:
Is the rule that the non-empty addrmaps are not allowed in the specification?
I think that the rule totally makes sense, don't get me wrong.

However I am doing some non-standard things with systemRDL and sometimes would like to have a component that just contains signals, so instead I use addrmap for this use.
I could patch your compiler to do that, but I don't want to fork it really.

A solution to my problem that might be useful for other things is to have a `-Wno-error=<error>` flag while compiling. I don't know if you see any other use for this flag

amykyta3, 2023-06-02:
Yes. The spec explicitly states that components shall all contain at least one structural instance (some exceptions like `mem` can be empty). An addrmap that only contains signals does not make sense, as it does not describe any addressable elements.

I do plan on enhancing the message system (#168) however I do not intend to provide a mechanism to suppress validation errors as it is essential to enforce these rules in order to guarantee assumptions other exporters make.