SystemRDL/systemrdl-compiler #178: Software-writable field spanning multiple words should cause error message

kjeldbeg, 2023-06-13:
Here is my example:

```
addrmap mymap_t {
  reg {
    accesswidth = 32;
    regwidth = 64;
    field {
      sw = rw;
      hw = r;
    } myfield[40];
  } myreg;
};
```
I expected it would fail to compile as the field is software writable and requires two accesses. As stated in the SystemRDL 2.0 spec. 10.6.1f:

<blockquote >
<ol type="a" start="6">
  <li>
Any field that is software-writable or clear on read shall not span multiple software accessible sub-words (e.g., a 64-bit register with a 32-bit access width may not have a writable field with bits in both the upper and lower half of the register).
  </li>
</ol>
</blockquote >

Is this check missing in the compiler?

jeras, 2023-11-17:
This seems an impractical requirement in the standard. The most common use case, where you would be in violation of the rule is having a 32-bit system bus and 64-bit counters timers. Writing the 64-bit register as a pair of HI/LO 32-bit registers is rather common, but it would also reduce practicality. In generated HDL code it would be nice to have 64 bit vectors without hand written glue logic, and in generated SW code it would be nice to be able to tell a 64-bit CPU, it can use 64-bit load/store operations.

amykyta3, 2024-12-13:
Indeed this rule has been relaxed for exactly the reason jeras stated above.

closed 2024-12-13 by amykyta3