SystemRDL/systemrdl-compiler #180: User defined property of list type isn't shown in get_property method

markcohen32, 2023-07-06:
**Describe the bug**
I'm trying to add a user-defined property of string[] or just string type and use it in one of my registers with its default value.
When I try to access it by using get_property(<udp>) I get 'None'

Details like these can be helpful:
property myUDP    { type = string ; component = reg; default = "STR";}; 

RDL code:
reg reg1 {
      myUDP;
};
reg1 reg1_instance;

I expect that if no value was assigned to this UDP, I would get "STR" as a result of the method.

amykyta3, 2023-07-19:
I confirm that the compiler throws an exception if a UDP with `type = string[]` is used. That should be supported and is definitely a bug.

However I do not see any issue with the non-array example you show. I do indeed see that with `type = string`, the property resolves as expected: `reg1_instance->myUDP == "STR"`

amykyta3, 2023-08-04:
Fixed in 1.26.1

closed 2023-08-04 by amykyta3