SystemRDL/systemrdl-compiler #185: Unintended `external` applied to `addrmap` instance

jeras, 2023-10-04:
Using v1.26.1 installed with pip.

Using the attached SystedRDL source file I get the `external` attribute set to `True` on the instances of GPIO `addrmap` (at the bottom of the file).
I did not specify anything as `external` explicitly, and I am unable to identify an implicit mechanism causing this (like user defined read/write side effects).

I use the attached file to test my code generator, and I am rather conservative with supported features, but I am still unable to find a reason why external would be applied.

The fastest way to reproduce this issue might  be to use the SystemVerilog code generator and check the `Example_pkg.sv` generated file.

[Example.txt](https://github.com/SystemRDL/systemrdl-compiler/files/12804777/Example.txt)

amykyta3, 2023-10-07:
This is intentional and is in accordance to the SystemRDL spec. See section 13.4.1-c.4:
> 4) `addrmap` instances are always considered external. An address with offset relative to the
beginning of the `addrmap` instance is provided.

jeras, 2023-10-07:
I have read a bit more about the distinction between `addrmap` and `regfile` in the standard. The simplest approach would be for me to use `regfile`.
In a case where I have multiple existing peripherals already defined separately as `addrmap` , is it possible to instantiate them as internal? I just tried, it is not. So, does this mean the best approach would be to define reusable peripherals with `regfile` and only wrap them in `addrmap` if I need to define some extra properties?

I know I probably answered my own question. I have been coding a generator all week, and I really missed somebody to brainstorm it with. It took me more than I expected, and I only got to the prof of concept level, I still have to add functionality, generalize, document, ... At the end I will write an issue, where I will list the problems I spent most time solving, so they could be added to the documentation as examples. I do not think my Python is good enough for contributions yet, but I could still try to contribute somehow.

amykyta3, 2023-10-08:
I would recommend the following: Any time you have a defined hardware RTL boundary like a peripheral's register block, use an `addrmap` as the top component. If you need to define additional internal hierarchical structures within that register block, use `regfile`. This is in-line with what the RDL spec vaguely suggests. I think of `regfile` components as conceptual groupings of registers rather than "true" structural hierarchies.
If you think about it that way, then an `addrmap` always being external makes a bit more sense - if you have a sub-hierarchy within a register block that is an `addrmap` then conceptually it is external to that register block. PeakRDL-regblock handles this situation by exposing an external interface to that sub-component.

Regarding documentation, any feedback is always very welcome! As the main developer of this project, I implicitly have a big blind-spot for what may not be completely intuitive to newcomers. If you have any ideas on how to improve the documentation, I would be happy to consider them! Beware that changing the Python API is harder since I need to honor the promise of compatibility to users.

closed 2023-10-09 by jeras