SystemRDL/systemrdl-compiler #186: missing parameterization in the component hierarchy

jeras, 2023-10-09:
I was writing a code generator for VHDL records representing the REDL hierarchy.
This requires all records for subbranches to be defined, before a record for a branch can be created.
I spent a day thinking about how to generate the records in the right order.
I was ready to store all records into a list and than reverse it, when I figured, I could just create the record on walker exit instead of using the walker entry callback.
I construct VHDL type names as `f"{node.inst.get_scope_path('_')}_{node.inst.type_name}"`.
I use a Python set type variable to check if I already defined a type, so there is no overhead, of duplicate types.
It would be much easier if there was an example doing this, for example creating C language structures.
In this case it would help to have an option to create filler nodes for the reserved/undefined fields/addresses, but this is material for a separate discussion.

It all seems to work well except for parameterization, which work only partially.

Here is the SystemRDL file I used for a showcase of this issue:
```SystemRDL
regfile street #(
    longint unsigned NUM = 32
){
    reg house {
        field {} INHABITED [NUM];
    } HOUSE;
    reg car {
        field {} ABANDONED [NUM-1:0];
    } CAR;
};

addrmap city {
    street              OAK;
    street #(.NUM ( 8)) BIRCH;
    street #(.NUM (16)) ACCACIA;
    street              SQUARE[4];
    street #(.NUM (16)) ROAD[4];
};
```

In this case, I get parameterized versions of `node.inst.type_name` for `regfile street`, but not for `reg house/car` and `field`s inside. The attached Pyhon code generates the next output:
```
$ ./issue.py city.rdl 
DEF-field: street_house_INHABITED [32-1 downto 0]

DEF-component: street_house
  INS-field: INHABITED [32-1 downto 0]

DEF-field: street_car_ABANDONED [32-1 downto 0]

DEF-component: street_car
  INS-field: ABANDONED [32-1 downto 0]

DEF-component: street
  INS-component:   HOUSE : street_house_Type
  INS-component:   CAR : street_car_Type

DEF-component: street_NUM_8
  INS-component:   HOUSE : street_house_Type
  INS-component:   CAR : street_car_Type

DEF-component: street_NUM_10
  INS-component:   HOUSE : street_house_Type
  INS-component:   CAR : street_car_Type

DEF-component: city
  INS-component:   OAK : street_Type
  INS-component:   BIRCH : street_NUM_8_Type
  INS-component:   ACCACIA : street_NUM_10_Type
  INS-com_array:   SQUARE : street[0:4-1]
  INS-com_array:   ROAD : street_NUM_10[0:4-1]
```

I do not understand the compiler well enough to understand whether this is a bug in the compiler, or I am doing something wrong, or I am just supposed to do something more.

I attached the RDL file and the Python generator.
[issue_city.zip](https://github.com/SystemRDL/systemrdl-compiler/files/12848734/issue_city.zip)

amykyta3, 2023-10-10:
This is working exactly as intended. The reason the `street` component gets its type name augmented by the parameter is because that is the component that the parameter definition is bound to. 
It is true that the child reg components `house` and `car` also have their contents augmented by the same parameter, but that is merely due to the availability of that parameter within the lexical scope. `house` and `car` themselves do not have any parameter bound to them directly.

Also due to how lexical scopes work, the `house` and `car` components do not need their type names to be augmented further since they are guaranteed to be distinct within that lexical scope.
The language reference has more details on how type names are augmented in section 5.1.1.4. This compiler also [extends this type name generation further](https://systemrdl-compiler.readthedocs.io/en/stable/dev_notes/dpa_type_generation.html) to cover situations where dynamic property assignments augment a named type.

Regarding how to generate unique & reusable structs - I do the exact same thing in several places and it is nearly identical to what you are currently doing:
* UVM class generator: https://github.com/SystemRDL/PeakRDL-uvm/blob/main/src/peakrdl_uvm/exporter.py#L231-L242
* Regblock hwif struct generator: https://github.com/SystemRDL/PeakRDL-regblock/blob/main/src/peakrdl_regblock/hwif/generators.py#L15
* C header struct generator: https://github.com/SystemRDL/PeakRDL-cheader/blob/dev/src/peakrdl_cheader/header_generator.py#L253-L256

jeras, 2023-10-10:
I tried `cheader` (installed with pip) and got a jinja related error, lets skip this one for now.
The UVM class generator is a bit different, since UVM classes can be parameterized, so they provide much more flexibility compared to SystemVerilog or VHDL-2008 RTL. Well Vivado documentation states SystemVerilog class support in synthesis, but this is again beyond this discussion.

So I will focus on the `regblock` generator. Again the one installed with pip. I am not sure how to check the version or whether it matters (whether there were related changes done to the sources.

I run the `regblock` generator:
```sh
peakrdl regblock city.rdl -o regblock/ --cpuif apb4
```
And I checked the generated SystemVerilog package `city_pkg.sv`.
It only contains the parameter default 32-bit field definitions, 16 and 7 bit field definitions are not generated.

If I try to generate markdown, just to check again if my interpretation of the standard and your response are correct:
```sh
peakrdl markdown city.rdl -o city.md
```
I can clearly see the 7 and 16 bit fields.

As a conclusion, I would say you have the same issue (I need a solution for) in the `regblock` generator.

jeras, 2023-10-10:
I had a look at the UVM generator and it does generate the extra parameterized types.
I will use the code as reference, but I have run out of time for this project, maybe I will get back someday.
Still, UVM classes could use their own parameterization features instead of making a new class for each parameterization.

amykyta3, 2023-10-10:
Oh I see what you were originally asking. I misinterpreted your question. I will check your original Python code to see what is missing.

jeras, 2023-10-10:
Thanks.

I might be able to do it myself, but I might not be as lucky as when I figured using the `exit` callbacks would generate definitions in the correct order of dependency. So I might write an ugly hack which would plague my company or/and the internet for eternity.

Seriously if time permits, I will try to repay you by writing a simple example of this problem for the documentation.
Separately I will write down all the research I have done on a user defined property for specifying field types:
- boolean,
- arbitrary precision integers,
- fixed point.

jeras, 2023-10-11:
I agree with the assessment that this would be a common usage pattern and that adding it to the API would help.

I mostly understand the distinction between the lexical scope a resolved type and the instantiation path, but when I have to use them properly, or describe them unambiguously, they become confusing (like pointers in C). I would like to think through where exactly in the hierarchy should be the parameterization, so I could comment on your implementation, but I would probably just add to the confusion. Still I will use the new API in my code.

amykyta3, 2023-10-12:
See new method added in v1.27.0: [Node.get_global_type_name()](https://systemrdl-compiler.readthedocs.io/en/stable/api/node.html#systemrdl.node.Node.get_global_type_name)

closed 2023-10-12 by amykyta3

amykyta3, 2023-10-12:
Glad it is working for you!

The extra code for fields and their width is to properly disambiguate type names in this situation:
```
field thing {};

thing thing1;
thing thing2[8];
```
Despite using the same component type declaration, I would consider these to be different types since they are different widths. `thing1` would use the original type name, and `thing2` would include a suffix of `_w8` since it was altered to a different bit width from the original definition.
I haven't decided yet, but in the future I may move this to be part of the extended type name generation described here: https://systemrdl-compiler.readthedocs.io/en/stable/dev_notes/dpa_type_generation.html

jeras, 2023-10-12:
I did not check what the standard would state regarding this code, and came up with this interpretation.

Instead of a single 8-bit vector field, this is an array of length 8 of single bit fields.

jeras, 2023-10-12:
Regarding dpa_type_generation, did you try to contact Accellera to ask the original authors for their interpretation.

I wrote them once regarding the old SystemC license, which was not open source enough for Debian. They did reply with a polite refusal, but the point is they did reply. Years later they did change the license, there were probably others asking about this.

My point is Accellera might be willing to give your contact information to some of the original authors and some might reply and be willing to discuss this issues with you.

amykyta3, 2023-10-12:
Regarding 8-bit vector for fields: This is clearly described in the spec.
See section 9.3

I have had several contacts from the SystemRDL committee that helped
greatly with the initial interpretation of the spec. Responses have been
mixed though. Some have been very helpful, others completely unresponsive.
I would love to see more engagement from Accellera regarding the RDL
standard.
I would recommend reaching out here:
https://forums.accellera.org/forum/52-systemrdl-discussion/
One of the committe members, Richard Weber, appears to be sparsely active
there.
There are also a few email aliases listed in the spec or other Accelera
pages. They may be tired of getting my emails, but perhaps if others
reached out they would respond :-)