SystemRDL/systemrdl-compiler #192: if struct is not fully defined, compiler crashes, it should point the line with to the error in the SystemRDL file

jeras, 2023-11-06:
I created a user defined property based on a custom structure:
```SystemRDL
// structure defining the data type
struct datatype {
    boolean          sign;
    boolean          fixedpoint;
    longint unsigned scale;
};

// user defined property
property datatype {
    type = datatype;
    component = field | signal;
};
```

I experimented with keeping parts of the structure undefined, in an attempt to use undefined as an option, I left some of the structure elements undefined. This caused a crash in the compiler:
```
$ ./issue.py test.rdl
Traceback (most recent call last):
  File "/home/ijeras/Cosylab/cslLib/regmap/datatype/./issue.py", line 80, in <module>
    root = rdlc.elaborate()
  File "/home/ijeras/.local/lib/python3.10/site-packages/systemrdl/compiler.py", line 402, in elaborate
    walker.RDLWalker(skip_not_present=False).walk(
  File "/home/ijeras/.local/lib/python3.10/site-packages/systemrdl/walker.py", line 159, in walk
    self.walk(child, *listeners)
  File "/home/ijeras/.local/lib/python3.10/site-packages/systemrdl/walker.py", line 159, in walk
    self.walk(child, *listeners)
  File "/home/ijeras/.local/lib/python3.10/site-packages/systemrdl/walker.py", line 159, in walk
    self.walk(child, *listeners)
  File "/home/ijeras/.local/lib/python3.10/site-packages/systemrdl/walker.py", line 150, in walk
    self.current_action = self.do_enter(node, listener)
  File "/home/ijeras/.local/lib/python3.10/site-packages/systemrdl/walker.py", line 176, in do_enter
    action = listener.enter_Component(node) or WalkerAction.Continue
  File "/home/ijeras/.local/lib/python3.10/site-packages/systemrdl/core/elaborate.py", line 42, in enter_Component
    node.inst.properties[prop_name] = prop_value.get_value()
  File "/home/ijeras/.local/lib/python3.10/site-packages/systemrdl/ast/literals.py", line 103, in get_value
    return self.struct_type(resolved_values)
  File "/home/ijeras/.local/lib/python3.10/site-packages/systemrdl/rdltypes/user_struct.py", line 106, in __init__
    raise ValueError("Cannot map 'values' to this struct")
ValueError: Cannot map 'values' to this struct
```

The standard states that all structure elements should be defined.
Section _6.3.2.3.2 Semantics for instantiated structs_ point _c_.

> All the members from a struct instance shall be assigned a value, either explicitly or by default.
Undefined struct members shall raise an error.

I think, the compiler should check if the structure is fully defined and issue an error if it is not.

I attached the code (Python script and RDL file, the last register triggers the issue) used to generate this error.
[issue_test.zip](https://github.com/SystemRDL/systemrdl-compiler/files/13272488/issue_test.zip)

amykyta3, 2023-11-07:
Agree this is a bug. Thanks for finding and reporting it!

amykyta3, 2023-11-07:
Added error message as per 6.3.2.3.2-c, and fixed missing default assignment behavior described in 6.3.2.3.2-b.
Fix will be in the next release.

Thanks again for exploring the tool and finding this bug!

closed 2023-11-07 by amykyta3