SystemRDL/systemrdl-compiler #195: Allow Elaboration without addrmap [FEATURE]

EAlexJ, 2023-12-06:
It would be handy to allow other nodes than addrmaps to be the top most nodes. At the moment, inputting a file that does not contain an addrmap throws a fatal error. This is limiting when it comes to generating output from regfiles definitions. For example creating headerfiles should not require the input of an addrmap.

I considered writing an importer that would allow this by wrapping the regfile in a dummy addrmap, but wanted to discuss this anyway. Is this the way a situation like mine is supposed to be handled?

EAlexJ, 2023-12-06:
I noticed that the definition does not represent an instance and I assume that an instance requires an addressmap.

This would mean that the importer is the way to solve this, as the compiler itself has no use for definitions only.

Is my assumption correct?

amykyta3, 2025-01-04:
This is unlikely to be something that can be supported on the compiler level. In many cases, an enclosing addrmap will provide additional contextual information that is necessary for RDL elaboration.
That said, it should be possible to extend some of the exporters to be able to export a specific sub-node. I can definitely see how that would be useful for C header generation. I have created a feature issue for the C header to support something like this: https://github.com/SystemRDL/PeakRDL-cheader/issues/19

closed 2025-01-04 by amykyta3