SystemRDL/systemrdl-compiler #202: Describe how addressing is impacted by ispresent in docs

hughjackson, 2024-02-06:
https://github.com/SystemRDL/systemrdl-compiler/blob/f9fc64029f1337cb30e984c84a15f07d8d842ea3/docs/dev_notes/inferred_placement.rst?plain=1#L83C1-L83C25

The default behaviour of the tool is to place implicit addressed component after components whose ispresent property is false. It is not clear from the systemRDL spec if this is correct, or if the addressing should behave as if the component was never specified at all. Either way, I think it would be useful to add a note to the above section describing the behaviour

amykyta3, 2024-02-06:
See the SystemRDL spec, Section 5.3.1-g:

*"Instance positions are computed presuming all instances are present.
Removing an instance canintroduce a hole."*

hughjackson, 2024-02-09:
Ah, I missed that somehow, thanks for pointing me at it!

closed 2024-02-09 by hughjackson