SystemRDL/systemrdl-compiler #221: Registers overlapping after elaborating model tree with compiler

Saacman, 2024-09-05:
Sorry for the long issue, I'm trying to provide as much context as possible.
TL;DR: When trying to elaborate a model tree, registers with sequential fields are reported to have a bigger size, resulting in offset overlap.

I'm trying to write a class to flatten a model tree from RDL files. This is what I have so far:
```python
#!/usr/bin/env python3

from systemrdl import RDLListener, RDLImporter, RDLCompiler, RDLWalker
from systemrdl.node import FieldNode, RootNode
from print_hierarchy import MyModelPrintingListener

# Define a listener that will print out the register model hierarchy
class Pruner(RDLListener, RDLImporter):
    def __init__(self, compiler: RDLCompiler):
        super().__init__(compiler)
        self.indent = 0
        self.component_stack = []

    def flatten_file(self, node):
        if not isinstance(node, RootNode):
            self.msg.fatal("Node must be a root node", self.default_src_ref)  
        walker = RDLWalker(unroll=True)
        walker.walk(node, self)

    # Listener methods
    def enter_Component(self, node):
        if not isinstance(node, FieldNode):
            self.indent += 1
    def exit_Component(self, node):
        if not isinstance(node, FieldNode):
            self.indent -= 1

    def exit_Addrmap(self, node):
        # print("Exiting addrmap", node.get_path())
        # Add the chidlren REGS to this component -> collect children reg from stack
        if self.indent == 1:
            addrmap_def = self.create_addrmap_definition(node.get_path())
                        
            _ = [self.add_child(addrmap_def, reg_inst) for reg_inst in self.component_stack]
            
            self.register_root_component(addrmap_def)
        

    def enter_Reg(self, node):
        #print("Entering register", node.get_path())
        # Create a REG definition
        reg_def = self.create_reg_definition()
        for field in node.children():
            # Field definition
            field_def = self.create_field_definition()
            # Assign properties to definition
            for field_property in field.list_properties():
                self.assign_property(field_def, field_property, field.get_property(field_property))
            
            # Instantiate the component definition
            field_inst = self.instantiate_field(
                field_def,
                field.get_path_segment(),
                field.lsb,
                field.msb - field.lsb + 1
            )
            
            self.add_child(reg_def, field_inst)

        # finish instantiation
        reg_inst = self.instantiate_reg(
            reg_def,
            "_".join(node.get_path_segments()[1:]),
            node.absolute_address *4
        )
        self.component_stack.append(reg_inst)
```
Maybe not the most efficient way to do it, but it seems to work. Except for this part in the enter_reg method:

```Python
reg_inst = self.instantiate_reg(
            reg_def,
            "_".join(node.get_path_segments()[1:]),
            node.absolute_address *4
        )
```
If I don't add the *4, The compiler produces the following errors:

```
error: Instance 'INTCTRL' at offset +0x1:0x4 overlaps with 'CTRL' at offset +0x0:0x3
error: Instance 'STATUS' at offset +0x2:0x5 overlaps with 'CTRL' at offset +0x0:0x3
error: Instance 'STATUS' at offset +0x2:0x5 overlaps with 'INTCTRL' at offset +0x1:0x4
error: Instance 'DATA' at offset +0x3:0x6 overlaps with 'CTRL' at offset +0x0:0x3
error: Instance 'DATA' at offset +0x3:0x6 overlaps with 'INTCTRL' at offset +0x1:0x4
error: Instance 'DATA' at offset +0x3:0x6 overlaps with 'STATUS' at offset +0x2:0x5
fatal: Elaborate aborted due to previous errors
```
I'm using the example files `atxmega_spi.rdl` and `tiny.rdl`. I nested the AddrMap a few times to try my PoC. The errors only occur with the atxmega registers, it seems as they increased 4 times in size. They are defined as follows:

```
reg {
        name = "Control Register";

        field {
            desc = "Controls the SPI clock rate when configured in master mode";
        } PRESCALER[1:0] = 0;

        field {
            desc = "These bits select the transfer mode";
        } MODE[3:2] = 0;

        field {
            desc = "Selects master mode when written to one, and slave mode when
            written to zero. If SS is configured as an input and driven low while
            master mode is set, master mode will be cleared";

            hw=rw; we;
        } MASTER[4:4] = 0;

        field {
            desc = "DORD decides the data order when a byte is shifted out from
            the DATA register. When DORD is written to one, the least-significant
            bit (lsb) of the data byte is transmitted first, and when DORD is
            written to zero, the most-significant bit (msb) of the data byte is
            transmitted first";
        } DORD[5:5] = 0;

        field {
            desc = "Setting this bit enables the SPI module. This bit must be
            set to enable any SPI operations";
        } ENABLE[6:6] = 0;

        field {
            desc = "When this bit is set, the SPI speed (SCK frequency) will be
            doubled in master mode";
        } CLK2X[7:7] = 0;
    } CTRL @ 0x0;
```

I'm not sure if I'm making any mistakes, any guidance would be helpful

amykyta3, 2024-09-06:
It looks like you are missing the regwidth property assignment.
By default, RDL assumes a register width of 32-bits, which means each register consumes 4 bytes of address space.

The atxmega example uses a default assignment to `regwidth` to set it to 8-bits for each register:
```
default regwidth = 8;
```
It looks like this was omitted in your test.

See the original file: https://github.com/SystemRDL/systemrdl-compiler/blob/main/examples/atxmega_spi.rdl#L9

Saacman, 2024-09-06:
Yeah that solved my issue, I forgot to consider copying the properties of the addrmap and regs.

closed 2024-09-06 by Saacman