SystemRDL/systemrdl-compiler #222: Registering Soft UDPs with an enum as the valid_type

rbroich, 2024-10-13:
I have a user defined property defined as follows in an RDL file:

```
enum E_FIELD_TYPE {
    FT_UNSIGNED {desc = "Unsigned integer number from 0 to 2^bits - 1";};
    FT_SIGNED {desc = "Signed number from -2^(bits-1) to 2^(bits-1) - 1";};
    FT_BOOL {desc = "Boolean value: true or false";};
};
property field_type {
    component = field;
    type = E_FIELD_TYPE;
};
```

A custom generator then makes use of this and everything works fine. After reading about Soft UDPs and the validation and unassigned default benefits they provide, I would like to register this property as a soft definition. Here is what I have tried so far:

```
class FieldTypeDefinition(UDPDefinition):
    name = "field_type"
    valid_components = {Field}
    valid_type = rdltypes.user_enum.UserEnum
```

This results in:
> error: The property definition for the feature extension 'field_type' uses a different 'type' definition from what this tool expects

Is there any way to do this? I have thought about defining the enum in a separate file and compiling it first (rdlc.compile_file) before registering the UDP (rdlc.register_udp), but then I don't have a way to reference it from inside the FieldTypeDefinition class. I suppose creating the enum inside the class could work, but the `existing_udp.valid_type != udp.valid_type` rulebook entry would still complain then.

Do you have any other suggestions?

amykyta3, 2025-01-03:
This is currently not supported, and would require some new concepts to be introduced into the compiler.
There is a problem of conceptual ownership here - the enum definition is normally something that is provided by the RDL source by the user, so its definition is implicitly not available prior to compile to load as a Soft UDP.

In your example, it looks like you are using an enumeration as a mechanism to further extend a field type. If you still want to use the enumeration, you could define the UDP as an `int` type, and that would accept the assignment.
Alternatively you can use a string type for the UDP as well.

rbroich, 2025-01-04:
Thanks. I ended up using a string type and that worked well. Thank you for the awesome work you do :)

closed 2025-01-04 by rbroich