SystemRDL/systemrdl-compiler #237: [BUG]

SanaUllahSultan, 2024-12-10:
I have reviewed all SystemRDL documentation and your compiler. I found no mechanism for generating interrupts while achieving the threshold value. 

For example:
I have a FIFO inside RAL, and any FIFO, like the FIFO inside the UART IP, will generate an interrupt when it contains more elements than the watermark level. 

I also didn't get any mechanism to specify the signal for my interrupt field. I also want some other signals other than interrupt, which just show the status of my counter and other conditions.

I am also not clear about the next property in a field, why it is, and what its use case

If you have any solution on those please share with me.

amykyta3, 2024-12-11:
The SystemRDL language only describes the software/hardware interface layer, and not much beyond it. Larger structures like FIFOs cannot be described with SystemRDL alone, and would require external HDL logic to construct this behavior.
Similarly, a watermark threshold would be better to implement outside the scope of SystemRDL and generate a boolean bit that sets/clears the interrupt as appropriate.

SanaUllahSultan, 2024-12-11:
Good cover.

let's make another example for the counter field in SystemRDL. I want to make interrupt on-threshold accrue. hw = na, sw = wr than how can i implement this feature

amykyta3, 2024-12-11:
Many of these concepts are described in the SystemRDL specification. Please refer to that for more details.

Here is an example that should give you context on all three of your questions:
```

reg {
    desc = "Controls interrupt threshold of the counter";
    field {
        sw=rw; hw=na;
    } value[31:0] = 0;
} counter_threshold;

reg {
    desc = "Counter that increments from an external increment control";
    field {
        sw=r;
        hw=na;
        counter;
        incr;
    } my_counter[31:0] = 0;
} counter_register;

// Define a dynamic threshold for the counter
counter_register.my_counter->incrthreshold = counter_threshold.value;

reg {
    // Option 1: sticky bit that software must explicitly clear
    field {
        sw=rw;
        hw=na;
        intr;
        stickybit;
        woclr;
    } counter_ifg_sticky = 0;

    // Option 2: interrupt is not sticky and is level sensitive to the threshold
    field {
        sw=r;
        hw=na;
        intr;
    } counter_ifg_nonsticky = 0;
} interrupt_flags;

// Set the sticky flag if the counter was exceeded
interrupt_flags.counter_ifg_sticky->hwset = counter_register.my_counter->incrthreshold;

// set the flag directly based on the threshold
interrupt_flags.counter_ifg_nonsticky->next = counter_register.my_counter->incrthreshold;
```

closed 2024-12-28 by amykyta3