SystemRDL/systemrdl-compiler #238: write singal bit of interrupt register

SanaUllahSultan, 2024-12-10:
I need to know how I can clear 1 bit of the interrupt register on read or write. That will affect the other interrupt fields on the same register address. 

For example:
register_1 = crc_intr, power_intr, tx_intr, rx_intr
If I read rx_intr to clear it, then I have to read register_1  if there is more than one rclr property of the interrupt field, then it will clear all. I need to know how SystemRDL handles those issues with respect to RTL and address decoding.

amykyta3, 2024-12-11:
For interrupts, I usually prefer using actions that can be atomically performed by software.
One option is to make the interrupt flag have "write-1-to-clear" behavior:

```
field {
    sw=rw; hw=w;
    woclr;
    intr;
    stickybit;
} flag;
```

SanaUllahSultan, 2024-12-11:
great,

but you miss one thing: that field is part of the register, which contains a few more interrupt fields. If a generate write transaction on the flag field, then it also affects other fields. 
for example, reg1 bit 2 is an interrupt bit; I will generate a write on reg1 with the value of "reg1 | (1<<2)" I do not change other bits, but it affects the other field.
can you please comment on that

amykyta3, 2024-12-11:
If you make *all* interrupt fields write-one-clear, then they will not be affected by writes unless the software explicitly clears the interrupt by writing a 1 to that bit position.

This is a very common way to implement interrupt registers that are thread-safe for software.

SanaUllahSultan, 2024-12-11:
I think there will be some mechanism to access every security bit indecently. Its designer wishes to make his interruption bit clear mechanism. 

One thing the SystemRDL team can improve is in the next release or develop the best practice document.

closed 2024-12-28 by amykyta3