SystemRDL/systemrdl-compiler #240: how to derefece the proprty of a field

SikoVerilog, 2024-12-16:
swwe_ref->swwe = we->reset;

sewer_ref field swwe property depends on the we field reset property. I am getting this on print in Python
swwe  ==  <PropRef_reset sw_access.swwe.we->reset at 0x102322060>

Can you please help me to dereference this and extract the name of the reference field and its property name too in python

amykyta3, 2024-12-17:
If a property is a reference to another property, it is represented using the `PropertyReference` object.
See the documentation: https://systemrdl-compiler.readthedocs.io/en/stable/api/types.html#systemrdl.rdltypes.references.PropertyReference

From there, the `name` and `node` members of that class should help get you what you want.

SikoVerilog, 2024-12-17:
Thanks for sharing PropertyReference
can you please help the property with the custom condition like
swwe_ref->swwe = we->reset == 0;
compiler through an error on that.

amykyta3, 2024-12-17:
Please provide details including code samples and errors. Difficult to help without proper context

amykyta3, 2024-12-28:
This is not supported. The SystemRDL language spec from Accellera does not describe the semantics for how property references are supposed to work in this situation.

closed 2024-12-28 by amykyta3