SystemRDL/systemrdl-compiler #245: [BUG] December 2024 commits broke SystemRDL builds

paul-demo, 2025-01-06:
The changes from 1.27.3 -> 1.29.0 of systemrdl-compiler broke builds due to assertions failing in the commit here:
https://github.com/SystemRDL/systemrdl-compiler/commit/0e1dfac7b54f098cc446bfe8fa3e2422e8798ac1 

Details like these can be helpful:
```bash
Traceback (most recent call last):
  File "/some/path/python/3.9.17/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/some/path/python/3.9.17/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File /some/other/path/lib/python3.9/site-packages/peakrdl/__main__.py", line 3, in <module>
    main()
  File /some/other/path/lib/python3.9/site-packages/peakrdl/main.py", line 183, in main
    options.subcommand.main(importers, options)
  File /some/other/path/lib/python3.9/site-packages/peakrdl/subcommand.py", line 154, in main
    self.do_export(top, options)
  File /some/other/path/lib/python3.9/site-packages/peakrdl_regblock/__peakrdl__.py", line 207, in do_export
    x.export(
  File /some/other/path/lib/python3.9/site-packages/peakrdl_regblock/exporter.py", line 157, in export
    DesignValidator(self).do_validate()
  File /some/other/path/lib/python3.9/site-packages/peakrdl_regblock/validate_design.py", line 32, in do_validate
    RDLWalker().walk(self.top_node, self)
  File /some/other/path/lib/python3.9/site-packages/systemrdl/walker.py", line 159, in walk
    self.walk(child, *listeners)
  File /some/other/path/lib/python3.9/site-packages/systemrdl/walker.py", line 159, in walk
    self.walk(child, *listeners)
  File /some/other/path/lib/python3.9/site-packages/systemrdl/walker.py", line 150, in walk
    self.current_action = self.do_enter(node, listener)
  File /some/other/path/lib/python3.9/site-packages/systemrdl/walker.py", line 176, in do_enter
    action = listener.enter_Component(node) or WalkerAction.Continue
  File /some/other/path/lib/python3.9/site-packages/peakrdl_regblock/validate_design.py", line 47, in enter_Component
    if not ref_is_internal(self.top_node, value):
  File /some/other/path/lib/python3.9/site-packages/peakrdl_regblock/utils.py", line 59, in ref_is_internal
    if current_node.external:
  File /some/other/path/lib/python3.9/site-packages/systemrdl/node.py", line 789, in external
    assert self.inst.external is not None
AssertionError
```

- [x] I have reviewed this project's [contribution guidelines](https://github.com/SystemRDL/systemrdl-compiler/blob/main/CONTRIBUTING.md)

paul-demo, 2025-01-06:
Apparently this is not actually guaranteed after elaboration as the comment says:

[image]

amykyta3, 2025-01-06:
> 0e1dfac

Thanks! Can you provide a minimal rdl example that reproduces this? I must
have a testcase gap

paul-demo, 2025-01-06:
Looks like it's due to `hwclr` usage:

Short example:
```SystemRDL
// Minimal example to demonstrate issue:
// https://github.com/SystemRDL/systemrdl-compiler/issues/245

default regwidth = 32;
default accesswidth = 32;
default addressing = compact;
default littleendian = true;
default reset = 0;
default sw = r;
default hw = w;

signal {
    sync;
    activehigh;
} latch_stats;

regfile foo {
    default hw = na;
    default counter;

    // Issue appears to be related to including this line:
    default hwclr = latch_stats;

    reg {
        field {} cnt[31:0];
    } num;
};

addrmap my_csr {
    foo bar[16] %= 0x100;
};
```

amykyta3, 2025-01-06:
Thanks! Will fix tonight

amykyta3, 2025-01-07:
Turns out this was only a partial fix. Re-opening.
I yanked the problematic releases from Pypi in the meantime.

closed 2025-01-09 by amykyta3