SystemRDL/systemrdl-compiler #246: unannounced breaking change in 1.29.0

matt-dbs, 2025-01-07:
We have a private [PeakRDL](https://peakrdl.readthedocs.io/en/latest/) exporter that uses some underlying APIs from systemrdl-compiler. After a recent update, we started seeing errors that boiled down to:

```
AttributeError: 'MemNode' object has no attribute 'fields'
```

Bisecting this across releases, the relevant change seems to have occurred between systemrdl-compiler 1.28.0 and 1.29.0. I can’t easily bisect this situation down to the commit level, but fb2d98b8ddbf8d25cfb33a1201a7bb1ab2fea337 looks like a pretty reasonable candidate for the cause of this. From the commit message maybe it’s meant to relate to this changelog entry?

> `Node.children()`, `Node.signals()`, `Node.Registers()`, `RegNode.fields()`, `RegNode.aliases()`, and `FieldNode.aliases()` now return lists rather than generators.

But I’m not really sure, because the changelog entry alone to me would indicate no updates needed to be made to our code (it’s doing something like `for f in node.fields: …`).

Am I interpreting this correctly? I can study the commits themselves to figure out how to update our code, but it’s not obvious to me from the changelog. To be clear, this is not blocking us; we’ve just pinned to `systemrdl-compiler==1.28.0` for now. Ironically it looks like the explicit field gap stuff that’s being introduced would actually be useful to us.

amykyta3, 2025-01-07:
Sorry for the unexpected change! I did indeed miss adding it to the changelog. Will correct that.

This was actually somewhat of a bug in the tool that I had fixed, and it looks like you were implicitly relying on.
Originally, the `.fields()` utility method was implemented as a general utility for all node types. This is somewhat inappropriate since the only node that will ever have fields as its immediate children is a `RegNode`. In all other cases, `.fields()` would have always returned an empty list. The change was to only allow `.fields()` to be used on a node that is a `RegNode`

In your case, I would recommend changing the usage to check whether using `fields()` iteration is even valid for the given node:
```python
if isinstance(node, RegNode):
    for f in node.fields:
       ...
```

matt-dbs, 2025-01-07:
Thanks for the speedy reply! Just checked that this indeed fixes things on our side. Thanks for explaining the situation.

closed 2025-01-07 by matt-dbs