SystemRDL/systemrdl-compiler #247: clearing fields

cuber2116, 2025-01-19:
addrmap my_addrmap {
    reg target_register { regwidth = 32; accesswidth = 32; // The register to be cleared
        field {
            name = "target_field";
            sw = rw;       // Read-only for software
            hw = rw;      // Read-write for hardware
            reset = 0x0;
        } target_field[1];      // 32-bit field
    };

    reg  w1c_register{  regwidth = 32; accesswidth = 32; // The W1C register
        field {
            name = "w1c_another_field";
            sw = rw;      // Write-1-to-Clear for software
            hw = rw;        // Read-only for hardware
            reset = 0x0;
            //onwrite = target_register.target_field-> woclr;
        } w1c_another_field[1];      // 32-bit field
    };
    
    target_register tar_reg @0x0;
    w1c_register w1c_reg  @0x4;
};

Every time a 1 is written in the w1c_another_field, the target_field gets cleared.

How do I achieve this?

amykyta3, 2025-01-22:
Assign a reference to the trigger field to the target's `hwclr` property.
Then using the `singlepulse` you can make the trigger auto-clear itself.

```
addrmap my_addrmap {
    reg target_register {
        field {
            sw = rw;
            hw = rw;
        } target_field[1] = 0;
    };

    reg  w1c_register{
        field {
            sw = rw;
            hw = na;
            singlepulse; // Always self-clears to 0
        } w1c_another_field[1] = 0;
    };

    target_register tar_reg @0x0;
    w1c_register w1c_reg  @0x4;

    // Clear the target if trigger register is 1
    tar_reg.target_field->hwclr = w1c_reg.w1c_another_field;
};
```

cuber2116, 2025-01-24:
Thank you so much!
I also wanted to ask if there was a way to have to an interrupt signal which will be the bitwise OR of all the interrupt fields?

amykyta3, 2025-01-25:
SystemRDL natively supports interrupts and the implicit OR you describe. See section 17 of the [SystemRDL 2.0 spec](https://www.accellera.org/images/downloads/standards/systemrdl/SystemRDL_2.0_Jan2018.pdf). They have some pretty good examples of how interrupts work.

Samrat-03, 2025-02-04:
I have a question. In the example code, the implicit OR is done by "global_int.global_int->next = master_int->intr;" but in my case there are no global or master interrupt registers. I just have a few lower-level leaf interrupts and a top-level interrupt signal. So, can I use the same method for interrupt signal? Something like - interrupt_signal -> next = leaf_int -> intr.

cuber2116, 2025-02-14:
@amykyta3 When do we use `we` and `wel` in RDL? In the above example it is not being used.

amykyta3, 2025-02-27:
@Samrat-03 
The example does show interrupt chaining, but that is not always necessary. Simply having registers that contain `intr` fields will produce a top-level interrupt signal.

@cuber2116 
`we` and `wel` are useful if you want hardware to explicitly load a new value into a register. These properties provide a write-enable input signal that is used to control *when* hardware changes a value.

cuber2116, 2025-02-27:
Ah! Right, thank you for clearing that! @amykyta3

closed 2025-03-08 by amykyta3