SystemRDL/systemrdl-compiler #251: Usage of hwenable Property

vignajethkk, 2025-02-04:
Hey,
     I am trying to use `hwenable`  in one of the field. I was thinking the way i use `hwenable` was incorrect but later i found an example from [https://github.com/orgs/SystemRDL/discussions/149#discussioncomment-4941497](url) like below.

```
signal {
    activehigh;
} cfg_mode;

reg {
    field {
        sw=rw;
        hw=r;

        // clear if cfg_mode == 1
        hwclr = cfg_mode;
        // Ensure only bits [3:2] get cleared
        hwenable = 4'b1100;
        // Ensure that the hwclr action takes precedence over software writes
        precendence = hw;
    } cfg[3:0] = 0;
} reg_cfg;
```

The above example when compiled throws the below error

```
  File "<path>/.venv/lib/python3.12/site-packages/systemrdl/core/ComponentVisitor.py", line 79, in visitComponent_body
    self.apply_local_properties()
  File "<path>/.venv/lib/python3.12/site-packages/systemrdl/core/ComponentVisitor.py", line 792, in apply_local_properties
    rule.assign_value(self.component, prop_rhs, prop_src_ref)
  File "<path>/.venv/lib/python3.12/site-packages/systemrdl/properties/bases.py", line 99, in assign_value
    self.env.msg.fatal(
  File "<path>/.venv/lib/python3.12/site-packages/systemrdl/messages.py", line 242, in fatal
    self.message(Severity.FATAL, text, src_ref)
  File "<path>/.venv/lib/python3.12/site-packages/systemrdl/messages.py", line 191, in message
    raise RDLCompileError(text)
    
systemrdl.messages.RDLCompileError: Incompatible assignment to property 'hwenable'

The above exception was the direct cause of the following exception:
```

any idea about this error ? and the version of systemrdl-compiler i use is `1.28.0`

cuber2116, 2025-02-05:
Hey, hwenable is of reference type, you cannot assign any value to it.
Assign a reference to it maybe a signal like this

`signal {
    activehigh;
} cfg_mode;
signal {
    activehigh;
} cfg_mode_hw[4];

reg {
    field {
        sw=rw;
        hw=r;

        // clear if cfg_mode == 1
        hwclr = cfg_mode;
        // Ensure only bits [3:2] get cleared
        // Ensure that the hwclr action takes precedence over software writes
        precedence = hw; 
    } cfg[3:0] = 0;
} reg_cfg;
reg_cfg.cfg -> hwenable = cfg_mode_hw;`

vignajethkk, 2025-02-05:
yeah, i agree 
probably the example here https://github.com/orgs/SystemRDL/discussions/149 is outdated

closed 2025-02-05 by vignajethkk

amykyta3, 2025-02-27:
Ha! looks like I forgot the rules for `hwenable`. thanks for pointing that out.