SystemRDL/systemrdl-compiler #256: Do `children()` and `descendants()` return a deep copy?

jeras, 2025-03-02:
I tried to compare nodes and I noticed separate calls to `children()` do not return the same objects. Are they deep copies? Is there a specific reason for this implementation?

As an example, two consecutive calls return nodes with different addresses.

```
            print(node, node.children())
            print(node, node.children())
```

```
<AddrmapNode example32 at 0x78aa3d9a1d00> [<RegfileNode example32.tests at 0x78aa3c3d51c0>, <RegfileNode example32.uart0 at 0x78aa3c3d7590>, <RegNode example32.conf at 0x78aa3c3d6ba0>, <RegNode example32.ctrl at 0x78aa3c17d370>, <RegNode example32.stat at 0x78aa3c17c530>, <RegNode example32.ufixed at 0x78aa3c17c380>]
<AddrmapNode example32 at 0x78aa3d9a1d00> [<RegfileNode example32.tests at 0x78aa3c208f80>, <RegfileNode example32.uart0 at 0x78aa3c550ef0>, <RegNode example32.conf at 0x78aa3c3d51c0>, <RegNode example32.ctrl at 0x78aa3c3d6ba0>, <RegNode example32.stat at 0x78aa3c233ef0>, <RegNode example32.ufixed at 0x78aa3c17c920>]
```

jeras, 2025-03-02:
I had a look at the source code. Let me start by saying I do not have experience with factories so my interpretation of the code is as it is.

Apparently the component hierarchy is static while the node hierarchy is created dynamically during a call to `children()`.

Is there a reliable (preferred) method for comparing two nodes. I would like to know if the two nodes are at the same position in the node tree. I probably could compare paths, and at least for my case, comparing the absolute address seems a good choice. Still is there a preferred way for two nodes are the same node?

jeras, 2025-03-02:
I found it, https://systemrdl-compiler.readthedocs.io/en/stable/api/node.html#systemrdl.node.Node.__eq__

closed 2025-03-02 by jeras

amykyta3, 2025-03-03:
Yes using the `==` operator is the preferred way to check node equivalence.