SystemRDL/systemrdl-compiler #262: [FEATURE] Iterate over properties

nealcrook, 2025-04-22:
- [x ] I have reviewed this project's [contribution guidelines](https://github.com/SystemRDL/systemrdl-compiler/blob/main/CONTRIBUTING.md)

In my RDL I have a set of user-defined properties (approximately 5) some are associated with fields and some are associated with regs. I want these to end up in the views exported from RDL (eg, to appear in a derived UVM register model)

I'd like some mechanism that allows my exporter to iterate over all properties on a node. I don't think this is currently possible. Is it a mad/bad idea? When iterating I might need to skip some "standard" properties like reset value but then I could process all of the others without needing pre-knowledge of their names.

Currently, the only solution I see is for my exporter to use get_property for each of the user-defined properties in turn. This requires hard-wiring the names of the user-defined properties into the exporter, which will cause painful maintenance if I add a new one.

(I have read about pre-registering a UDP and about soft UDP. I understand the reasons for these mechanisms and how soft UDP (in partucular) provide some nice protection for the user and the back-end consumer of the exported views. In my particular application, the RDL is acting as an intermediate format exported from a proprietary database and the syntax of the RDL is "guaranteed by design")  

Thank you for your consideration

Neal.

amykyta3, 2025-04-22:
Hi Neal,
If I'm understanding your request correctly, I believe the mechanism to do what you want to do already exists.
See the [Node.list_properties()](https://systemrdl-compiler.readthedocs.io/en/stable/api/node.html#systemrdl.node.Node.list_properties) method.
This will return a list of all the property names that are valid for the current node type. Furthermore, the method has options to filter them down based on whether they were explicitly assigned, they are UDPs or they are native RDL properties.

Using that list, you can then call `Node.get_property()` individually.

Hope that helps!

nealcrook, 2025-04-23:
Hi @amykyta3,
thanks for your super-fast and courteous response. Yes, it sounds like this does exactly what I want, and the method's options also sound useful. I missed it in the documentation because I searched for "property" but not "properties".

Suggestion: it might be beneficial to mention get_properties() in https://systemrdl-compiler.readthedocs.io/en/stable/properties.html

thank for your help

Neal.

closed 2025-04-23 by nealcrook