SystemRDL/systemrdl-compiler #276: RDLImporter.assign_property does not work for array types

amykyta3, 2025-07-09:
### Discussed in https://github.com/orgs/SystemRDL/discussions/275

<div type='discussions-op-text'>

<sup>Originally posted by **engdoreis** July  9, 2025</sup>
Given the UDP:
```rdl
property tags {
  type = string[];
  component = field | reg;
};
```
The function `assign_property`
```py
  self.importer.assign_property(field, "tags", ["foo", "bar"])
```
Returns the error:
```sh
systemrdl.messages.RDLCompileError: Incompatible assignment to property 'tags'
```

Similarly, I could not find the type expected for the property `swwe` when it should reference another register field, I tried string, component, but all returne error.</div>

amykyta3, 2025-07-09:
Looks like there is some internal inconsistency regarding expectations on what `PropertyRule.assign_value()` expects.
In some cases it accepts post-elaboration literals, but does not handle the case where the value is an array (likely expects arrays to be `ast.ArrayLiteral` objects.

Clean this up. Investigate which makes more sense - either:
- Make `PropertyRule.assign_value()` support all literal types, including arrays
- Simplify `PropertyRule.assign_value()` by requiring **only** AST nodes. Wrap literals in AST at the importer layer.

Slight preference for the first option since it would limit the proliferation of useless AST wrapper objects (performance reasons). Generally prefer avoiding giving the elaborator more work wherever possible.

engdoreis, 2025-07-17:
Compounding this issue, the `encode` property exhibits the same bug; ideally, it will be resolved simultaneously.

amykyta3, 2025-07-17:
Enum encodes should work without issue. I use this mechanism in the IP-XACT importer. See:
* https://github.com/SystemRDL/PeakRDL-ipxact/blob/main/src/peakrdl_ipxact/importer.py#L952
* https://github.com/SystemRDL/PeakRDL-ipxact/blob/main/src/peakrdl_ipxact/importer.py#L673-L674

amykyta3, 2025-09-19:
Fix implemented. Will be in next release

amykyta3, 2025-09-19:
Release Candidate published. See: https://github.com/orgs/SystemRDL/discussions/289

closed 2025-09-22 by amykyta3

amykyta3, 2025-12-03:
References are not supported using the importer API yet (see #183).

This will likely require a more intrusive rewrite of how the importer API works. My initial implementation of it was a little shortsighted such that building valid reference objects would be cumbersome. I'm hoping to refactor this, as well as make the importer API generally easier to use.